Fluigent/fgt-SDK #97: PRM-FOEM blinking erratically , card not detected

JeremieBours, 2025-04-03:
Hello , 

For a client , we are integrating two cards in a machine :  
INT-FOEM4
PRM-FOEM 
Switch board 

The issue is : When using the software , 
the Switch board is detected 
The PRM-FOEM is **not detected  and blinking 5 times per seconds** , any idea what could we do ? 

The setup is similar to this : 
[image]

Using Python SDK 
Ubuntu 

We know that you are the software part of the team , but the Fluigent support team ask us to open an issue here, 

Best regard , 
Keep up the good work , we like your SDK very much.

Ygor-Oliveira, 2025-04-04:
Hello,

Could you use the log functionality to record what the instrument is communicating? Simply add a call to [`fgt_set_log_verbosity(5)`](https://github.com/Fluigent/fgt-SDK/blob/8463d0eb66d935caf5b8d2b9f5c195e2cdedc715/Python/Fluigent/SDK/__init__.py#L1148) at the start of your script (before any other calls to the Fluigent SDK) and run it. It should produce a .log file in the same folder you ran the script from. If you require confidentiality, you can send it to our support email, otherwise you can attach it to a comment here. The only potentially sensitive information the logs will contain is the instrument's serial number. If the log file is large, you can compress it to .zip, which should make it small enough to fit in a email attachment. Thanks in advance.

Best regards,
Ygor

Ygor-Oliveira, 2025-06-02:
Hello,

From the logs, the INT-OEM board responds as if there weren't any modules connected to it at all, so there is nothing that can be done on the SDK side. This indicates that your boards are damaged or improperly configured. Please contact support to discuss having them repaired or replaced.

Best regards,
Ygor